TomWhitwell/Magnetophon #5: MTM Magnetophon

boombipbass, 2018-07-30:
I just soldered together a Megnetophon, so far Ive succeeded the Mikrophonie and the Music thing,
I am sure I've checked the making the module manual quite close but it doesn't generate any sound from my tape loops I've stuck on a piece of wood with the 2nd head, also the 1st head is not responding nor do I get a feedback hum from other modules form the output.

How could I check with a multimeter if the module is getting power in the circuitry?

Best bbb

boombipbass, 2018-07-30:
Ok well Ive just learned that you should atleast never put a just soldered module in your Eurorack..

Oups haha. Well atleast Ive learned something on my own, but now to think, I do have a multimeter or are there better solutions?

Best bbb

TomWhitwell, 2018-07-30:
If the module is completely dead, check the op amp chip is getting power - look at the data sheet (google TL072) and see which pins need +v and -v. Put the black probe on -v and the red probe on +v and you should see around 23-24volts. Check the chip is correctly oriented.

boombipbass, 2018-07-30:
Is this the data sheet? https://github.com/TomWhitwell/Magnetophon/blob/master/Magnetophon_Schematic.pdf

How do I know which is 5+ and -6 of the 8 pins? Its faceing the right way aswell as the OpAmp Holder.

Also how can I check if the TapeHead/minijack male is working with my Multimeter?

Thanks for helping me ^-^

Love your stuff!

TomWhitwell, 2018-07-30:
It’a on page 5 here : http://www.ti.com/lit/ds/symlink/tl072.pdf

(NOT page 4, that’s the tl071)

boombipbass, 2018-07-30:
Ah yes by following the copper route on the backside I already found out that R17 connected with 7 and thus which 5 must be since that one connected to the OpAmp nr. 1 in the schematics, wow learning new things here 🗡

boombipbass, 2018-07-30:
Strange that a black probe must be on the - and a red on the + since the ribbon is otherwise in color :)

boombipbass, 2018-07-30:
is it AC of DC?

boombipbass, 2018-07-30:
Nope not with AC nor with DC its getting 23/24 volts. On my multimeter set to 20 V.
I've checked with the OpAmp holder aswell as with the OpAmp itself :- (

boombipbass, 2018-07-30:
I do have to say I havent put it in my modular system, because I read thats stupid and I should actuaklly use a Bench power supply or some sort?

boombipbass, 2018-07-30:
I am sure all connections are soldered good though

boombipbass, 2018-07-30:
[image]
[image]

boombipbass, 2018-07-30:
[image]
Also on modulargrid it says this about the magnetophon which i dont udnerstand?

TomWhitwell, 2018-07-30:
That just means nobody has put the power specs into modular grid - it draws around 25mA or less on each rail

boombipbass, 2018-07-30:
Ok thats good to know, then the only thing I am wondering, do I need a Bench power supply to test DIY kits or not?

And should I recheck the module in the case with the IC to see if it gets 24 volts, it has been in the case and no smoke or explosions so thats a good thing :)

boombipbass, 2018-07-30:
Ok well checked it with Power on and still no Votlage, any solution? :)

realitycontrol, 2018-07-30:
If you're measuring between the +V and -V pin on the OpAmp using the 20V DC setting on your multimeter (as stated above); you will not get a reading.

You want to be using the 200V setting when you are expecting anything over 20V.
If using the 20V setting you could separately measure the + and - pins to ground.

zerojardins, 2020-04-29:
As I can see on the pictures of your board, you misplaced both fuses with the two 100n capacitors (C4 & C5)

hope that helps