TomWhitwell/Magnetophon #8: Magnetophon playback

metalthrow, 2019-03-26:
I have a Magnetophon with an external playback head. When rubbing the tape head across tape stuck to a sheet of foamcore or hard paper, I've tried both, it just sounds like the back of a credit. I'm wondering if anyone has experimented with this and knows if it is possible to obtain the cassette's actual sounds on it when manually pushing the head on to the tape?

TomWhitwell, 2019-03-26:
Hi, Sorry to hear that. I've tested it both with cassette tape and with 1/4" tape from a Nagra. 
Clearly the 1/4" tape is much easier, because it's a single mono signal going one direction along the tape. 
However, I did get it to work roughly with cassette tape, although the signal was substantially lower in output, and your have four simultaneous tracks -  left & right x 2 (backwards & forwards) on each bit of tape, which can quite quickly become noise. 
A secondary issue was that when I tried it with spoken word tapes (language cassettes) I quickly realised that for every inch of speech there might be 3-4 inches of silence, which makes the whole thing a bit tricky. 
You're getting roughly 2 inches of tape per second of sound - which should give an idea how fast to move the head.

metalthrow, 2019-03-27:
Thank you for the reply so promptly. I'm going to try out some 1/4'' tape and see fi that helps. I also noticed in the demo video there something on top the tape , did you glue it to the paper? also does foamcore vs. paper make a difference