TomWhitwell/Magnetophon #9: Tape Wand

tetsugakusha66, 2020-02-26:
Hi Tom!
I'm having a big issue with my Magnetophon. 
The tape head in the module itself works all right, but as soon as I plug in anything into the Head In jack, it's noise city.
I soldered the tape head to a 3m long patch cable and when it's plugged in the ground noise is massive. I'm assuming this is a ground noise problem, because when I touch nearly anything on my desk, the noise is attenuated slightly. The tape head on the wand functions correctly so I don't think that I have soldered it in backwards.
Could this be due to the length of the tape head wand or the fact that I used a patch cable instead of a guitar cable?
I tried to populate the DNP spot with a 10nf cap, but to no avail.
Any advice you could offer would be greatly appreciated!

snoopyart, 2020-06-25:
Exactly the same issue for me. I also used a patch cable (high quality though, with shielding). Also tried the 10nf cap. Also no way to get it working with the wand. No idea. Checked all solder points etc. but the module itself works perfectly fine with the internal head... Any help appreciated! THX!

magnesi, 2020-07-23:
Same here. Any help would be hugely appreciated as I'm trying to use it for a college project.

TomWhitwell, 2020-07-24:
Re: ground noise problems for @magnesi @snoopyart @tetsugakusha66 
I connected mine using (from memory) a balanced cable - i.e. two internal connectors and one shield. The internal connectors went from tip/ring on the socket to the two pins on the tape head. The orientation doesn't seem to matter. The shield is connected at the socket end. At the tape head end I tried to connect it to the metal frame of the tape head, but I'm not sure this was super successful. I found it very hard to solder to the tape head body, so I stripped a bit of the shield and put it next to the tape head body, then held it in place with heat shrink. 
I don't if this will solve your problems, but it's worked OK for me.

magnesi, 2020-07-25:
Thanks Tom

esoterydactyl, 2022-06-11:
Very happy to have found this issue open, as I have been driving myself absolutely crazy this afternoon trying to understand what I've done wrong with the kit!  I'll see if I can work out a balance cable.  If this turns out to be the solution, we should consider updating documentation to recommend it!