TomWhitwell/Magnetophon #13: Very weak sound

jojo-monk, 2023-07-06:
Bonjour,

je viens de finir de construire le module, il fonctionne mais le son est vraiment très faible, même avec le gain au maximum.
J’ai remplacer le condensateur C6 par un 22nf au lieu de 15nf, car je n’avais que celui-ci sous la main, est-ce que ça peut être la cause du problème ?
Ou alors la tête de lecture ? J’ai utilisé celle-ci : https://fr.aliexpress.com/item/1005004626148845.html?spm=a2g0o.order_list.order_list_main.10.7c6b5e5bIA2tNH&gatewayAdapt=glo2fra

jojo-monk, 2023-07-06:
English traduction : Hello,

I just finished building the module, it works but the sound is really very low, even with the maximum gain.

I replaced the C6 capacitor with a 22nf instead of 15nf, because I only had this one on hand, can that be the cause of the problem?

Or the reading head? I used this one:

TomWhitwell, 2023-07-07:
Hi a few questions

1. What tape are you using to test the head - if you use a piece of commercial stereo cassette tape - there are 4 x channels at each point on the tape, overall each is very low. I've had best results from a mono 1/4 inch recording 
2. Are you using the tape head on the panel or on a separate wire / 'wand' - it's easier to get a good contact between tape and read with a separate head 
3. Is that a stereo tape head? I haven't used those - perhaps that has lower output than a mono head - maybe there is some way to wire the two channels together? 
4. I don't think the 22/15nf will make a big difference

jojo-monk, 2023-07-07:
I used stereo cassettes with a separate playback head, but indeed it is a stereo playback head, since it has 4 pins, I welded only 2, I will try to weld the 2 channels together, and try to recover 1/4-inch strips.

Thanks

jojo-monk, 2023-07-07:
J’ai testé en connectant les 2 canaux ensemble, et avec une bande plus large, c’est un peu mieux, mais le signal est encore trop faible… est-ce que ce n’est pas un problème d’impedance ? La tête que j’ai à une impédance de 260 ohm ? C’est peut-être trop… y a-t-il moyen d’adapter le module en changeant quelques résistances ou condensateurs ?

jojo-monk, 2023-07-07:
I tested by connecting the 2 channels together, and with a wider band, it's a little better, but the signal is still too weak... isn't it an impedance problem? The head I have at an impedance of 260 ohm? It may be too much... is there a way to adapt the module by changing some resistors or capacitors?

TomWhitwell, 2023-07-07:
I'm afraid I don't know - the schematic is here but I can't help beyond that